foxglove/studio #343: Docker container problem with foxglove-studio

goupil35000, 2021-03-24:
Hello,
I installed foxglove v0.2.0 in a docker container.
After installation of foxglove, I'm unable to use roslaunch or rosrun on every package.
Verified with a new container. Deinstallation of foxglove-studio doesn't resolve the problem.
Foxglove-studio seems to work correctly on the same computer directly outside the container.

$ rosrun turtlesim turtlesim_node
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-student'
[ INFO] [1616618619.246782155]: Starting turtlesim with node name /turtlesim
[ INFO] [1616618619.251715865]: Spawning turtle [turtle1] at x=[5.544445], y=[5.544445], theta=[0.000000]
dbus[6192]: The last reference on a connection was dropped without closing the connection. This is a bug in an application. See dbus_connection_unref() documentation for details.
Most likely, the application was supposed to call dbus_connection_close(), since this is a private connection.
  D-Bus not built with -rdynamic so unable to print a backtrace
Aborted (core dumped)

amacneil, 2021-03-24:
Hi @goupil35000 

Can you share your Dockerfile?

Also, I'm interested to hear why you want to run Foxglove inside a container. It's designed to run directly on your computer (although you can still connect to ROS running inside a container as long as you use `--network=host` mode to expose all the ports).

goupil35000, 2021-03-25:
Hi,
Thanks for your reply. 

Why using a container for us ? Because we all use the same container  and for security reason we are not allowed to install software outside our container. And in case of problem (for example this is the case after installing foxglove-studio), we launch a new container.

Our container is based at the beginning on osrf/ros:noetic-desktop-full, but many things were added to this without a Dockerfile. And the size of our container is now 25GB so impossible to send it.

So I retested it directly with osrf/ros:noetic-desktop-full. I updated softwares in container with apt and then add the foxglove deb file to the system. The problem is now different. I can launch turtlesim in this container. But foxglove is not working in the container.
$ foxglove-studio
Trace/breakpoint trap

amacneil, 2021-03-27:
I can reproduce this locally. We haven't tested running studio inside docker yet, so this isn't a fully supported workflow and several things could be at fault here.

```
ubuntu@d9938818b032:/$ foxglove-studio
Trace/breakpoint trap (core dumped)
```

If you're willing to get your hands dirty, the first thing to test would be whether you can launch an empty electron shell inside docker (try the same version of electron that we use). It sounds like electron is crashing, so studio might not be at fault here.

lucasw, 2021-04-28:
I have a dockerfile branched off of `main` https://github.com/lucasw/studio/blob/dockerfile/Dockerfile and branched off of `v0.6.0` https://github.com/lucasw/studio/blob/dockerfile_v0.6.0/Dockerfile

It's not getting through `yarn install` in either version, both fail with errors with spectron like

```
N0001: │ Error: spectron@https://github.com/foxglove/spectron.git#commit=7f88e8fe148ee14b3efc14c3646c08c4ed0a6747: Repository clone failed: spawn git ENOENT
```

As I understand it the `yarn start` actually launches a standalone electron application, it doesn't run in a browser and the docker container could just provide a server?  (Is there a youtube video showing the application getting launched and running?)  I think that can work with docker but of course the install needs to finish first before working on that.

amacneil, 2021-04-28:
@lucasw are you trying to develop studio inside docker or just to run it? If you just want to run it I would recommend using a release from https://github.com/foxglove/studio/releases rather than building from source.

If you want to run it from source, I can see two ways to achieve this:

1) Run the server (`yarn serve`) inside docker, and run the electron shell (`yarn start`) outside docker. That way the GUI will run on your host as expected. However, you would need to run `yarn install` and have all the node_modules files available to your host (outside docker), so at that point it would be easier to just run both processes outside the container (this is why we don't use docker for development, node is pretty self contained).
2) You could run both the server and electron inside the container. If you do this, you will need to set up X forwarding from your container to your host. Explaining how to do that is outside the scope of a github comment (it can be pretty complex depending on your setup). But it's certainly possible.

The original commenter on this ticket was trying to run a release build inside docker. That in theory should also be possible, you can `COPY` the `.deb` file into a docker image and apt install it. However, to make that work you would also need to set up X forwarding per (2) above.

Separately, the actual error you are getting above isn't related to any of what I just said. I think it looks like you just don't have git installed in your container. I would recommend using `FROM node:14` rather than manually installing node.js into `ubuntu:20.04`, it will make your Dockerfile a lot simpler.

lucasw, 2021-04-28:
yarn install `Repository clone failed: spawn git ENOENT` -> Dockerfile `RUN sudo apt install git` :heavy_check_mark: 

That's probably enough to get me going for now but here's the background:

I'd like to run it on vanilla Ubuntu 20.04 with Docker installed but minimal use of sudo, ppas, curl scripts and so on beyond that.

A Ubuntu snap or getting the deb upstreamed to an official repo is probably a big ask and great for others but not really what I would want: being able to use the latest version, a PR/forked version, or similar prior to a merge and release is really nice, and paves the way to minor contributions from me or anyone else coming from the ROS world with minimal experience in any of the technologies here.

amacneil, 2021-04-28:
Makes sense. 

Our team doesn't use a docker development workflow becuase (a) it's non-trivial to get X window forwarding working, and (b) our only real system dependency is node.js (v14+), and everything else remains contained within our folder (unlike developing on python, c++, etc).

So by far the easiest way to develop is to just install node.js on your Ubuntu machine and then you can work directly inside our folder and it won't pollute your machine.

However, developing in Docker isn't impossible. I do think it will be easier if your Dockerfile uses `FROM nodejs` though, and it will be easier to just bind mount our source directory rather than just copying files in. I'll make a PR to demonstrate.

amacneil, 2021-04-29:
Here is a working example of running `yarn serve` inside a docker container: https://github.com/foxglove/studio/pull/797

I haven't looked into running `yarn start` (electron) yet. Last time I tried to run X windows inside docker it was a rabbit hole, but maybe it is easier these days.

defunctzombie, 2021-06-01:
Closing this for now. Docker development workflow for the studio desktop app isn't something we are actively pursuing. If you have nodejs 14+ installed on your workstation getting started via `yarn`, `yarn serve`, and `yarn start` is what I would recommend as a workflow that aligns with how we develop the app and is most likely to work successfully. If someone does figure out how to do GUI dev in a docker container we would welcome some docs on the matter.

closed 2021-06-01 by defunctzombie