foxglove/studio #458: Can't connect to ROS

dignakov, 2021-04-03:
Stuck connecting when trying to open a connection to ROS on localhost (v0.3).

[image]

defunctzombie, 2021-04-03:
Few questions to help me debug this with you:

* What Operating System are you using?
* Is ROS running on your host machine or in a virtual machine? docker?
* Do you see any errors in any other part of the app?
* Could you open `View -> Advanced -> Toggle Developer Tools` and report and errors you see there.

Let's start with those and see if we learn anything new.

dignakov, 2021-04-03:
Thanks for looking into this. Happy to provide as much debugging info as I can, foxglove looks awesome and I'd love to use it.

OS: Ubuntu 20.04
ROS: Noetic, running locally
Foxglove: deb from the site

Error/Warnings/Output:

when launching foxglove
```
$ foxglove-studio 
APPIMAGE env is not defined, current application is not an AppImage
```

other terminal output:
```
[13729:0403/112908.175627:ERROR:CONSOLE(1)] "Session is unregistering, can't dispatch pending call to Debugger.setBlackboxPatterns", source: devtools://devtools/bundled/sdk/sdk.js (1)
[13729:0403/112950.161437:ERROR:CONSOLE(6)] "Uncaught TypeError: e.consoleMessage is not a function", source: devtools://devtools/bundled/console/console.js (6)
```

developer console (sorry, I'm not sure how to copy that so I took a screenshot instead):

[image]

Hope this helps.

amacneil, 2021-04-03:
Interesting. Is that the last error in the developer console if you scroll down? Usually when it is connecting, we see a message each time it attempts a connection. 

Do you see any other messages in the console when you try a rosbridge/websocket connection?

dignakov, 2021-04-03:
Sorry, I think it was the last error, but now I'm not sure...

I actually can't connect to websockets either. I opened a second issue for that (#459). But this is what the console displays before any connections and after attempting to connect to a websocker or ros. Nothing after this comes up on the console if I leave it alone and it just shows the wheel as it's trying to connect.

start:
[image]

websocket:
[image]

ros (closed foxglove, reopened and tried connecting to ros):
[image]

defunctzombie, 2021-04-03:
@dignakov I suspect I've traced down what the issue might be - and we have a fix in our main branch which has not made it into a release yet. We will release a patch version this weekend (latest Monday) for you to try.

amacneil, 2021-04-04:
@dignakov could you test this version?

https://storage.googleapis.com/foxglove-releases/v0.3.0-nightly.20210404.3dc0144/foxglove-studio-0.3.0-nightly.20210404.3dc0144-linux-amd64.deb

dignakov, 2021-04-04:
Still can't seem to connect.

start:
[image]

connecting to ros:
[image]

connecting to websocket:
[image]

defunctzombie, 2021-04-04:
@dignakov if you'd like to jump into our slack https://github.com/foxglove/studio#support I can get some more details there and try a few additional troubleshooting items.

defunctzombie, 2021-04-20:
Closing this. We've fixed some more issues with the app connections and startup. Let me know if this continues to not work for you.

closed 2021-04-20 by defunctzombie