foxglove/studio #495: Can't load panels

iceuw, 2021-04-07:
System: Windows 10
Describe: After launch program, all panels disappear after a flash
[image]

amacneil, 2021-04-07:
That's very strange. Which version are you using? Can you try v0.4.0 we released today?

iceuw, 2021-04-07:
I tried v0.4.0. And then there is nothing remains in the program window. And I tried the v0.3.0 with the Console in the Debug. It shows this error
[image]

iceuw, 2021-04-07:
And I found this file called "app.asar". But the "app.asar" is a file ranther then a directory. Is there any wrong?

amacneil, 2021-04-07:
Ok thanks. No, the `app.asar` is just how we package the app, and those `failed to load SourceMap` warnings you see are also normal and not causing the error.

I think I have an idea what is causing this, let me get back to you with troubleshooting steps.

thoys, 2021-04-09:
I've got the same issue myself

amacneil, 2021-04-09:
Are either of you using a non-English language setup in Windows? We're seeing some `DWriteFont::Create` errors which might be related, but unclear what the cause of them is.

closed 2021-04-13 by jhurliman