foxglove/studio #497: Expected 1 top level type definition for 'some_topic' but found 2

yairBAR-caja, 2021-04-07:
Hi,

1. Love foxglove, it is awesome!
2. im having issues when connecting to my robot via websocket. when i connect and all of the nodes load foxglove shows an error:
Expected 1 top level type definition for 'BaseRecoverResult' but found 2

BaseRecoveryResult.msg is made up of a bool field and a string field. this msg is used in a result topic of an action server.

here is a screen snap:
[image]

anyone know whats up?

defunctzombie, 2021-04-07:
@yairBAR-caja Do you possibly have two nodes publishing different versions of BaseRecoveryResult?

yairBAR-caja, 2021-04-08:
Yes you were right...

The action for base recovery auto-magically generates a msg with the same.
After changing the name of my custom msg the issue is resolved.

Thanks,
YAir

closed 2021-04-08 by yairBAR-caja