foxglove/studio #537: Feature request: Display Theora-encoded image frames

rgov, 2021-04-09:
The [theora_image_transport](http://wiki.ros.org/theora_image_transport) package provides the only image_transport plug-in that uses a video codec ([Theora](https://www.theora.org)) to drastically reduce the size of consecutive frames, which has an immense impact on rosbag file size -- in a recent test, it reduced the size of a rosbag by 99.7%. 

However this comes with usability tradeoffs because tooling is not able to decode the Theora stream. This includes Foxglove, which cannot display Theora frames in an Image panel.

Essentially, you would need to scan backwards from the current time to find the previous keyframe, then decode all frames between the keyframe and the current frame. You may need to find the initial header packets, as well.

You could possibly use [ogv.js](https://github.com/brion/ogv.js/). If Foxglove is based on Chromium, Theora may be supported already.

amacneil, 2021-04-09:
Makes sense. Is it possible to share a short bag file with theora-encoded topics? (you can send to us privately if needed).

rgov, 2021-04-12:
Sure, here is a bag with 50 `theora_image_transport/Packet` messages on the topic `/camera_array/cam0/image_raw/theora`.

I researched the Theora specification [here](https://theora.org/doc/Theora.pdf) (PDF). Here is an important bit:

> Before decoding can begin, a decoder MUST be initialized using the bitstream headers corresponding to the stream to be decoded. Theora uses three header packets; all are required, in order, by this specification. Once set up, decode may begin at any intra-frame packet ... belonging to the Theora stream. In Theora I, all packets after the three initial headers are intra-frame or inter-frame packets.

So, you would scan the topic to find these 3 headers. Naively, they would be the first 3 packets on the topic, but more on this in a second. Their `data` fields should start with the octets `0x80`, `0x81`, and `0x82` respectively.

Every other packet should be a video frame that can be decoded to an image. There are two kinds, "intra" frames (like keyframes, they are a complete image) and "inter" frames (like a delta from the previous frame). To identify these video frames, check the first two bits. `0b00` is an "intra" keyframe and `0b01` is an "inter" keyframe.

N.B.: I don't think you should assume that a Theora topic contains *only* a single video. For instance, the bag might capture multiple executions of a node that outputs video, and it will output a new header for each execution. So you would probably want to keep a lookup table of all the header packets in the bag, and re-initialize the decoder any time playback jumps over one of the boundaries.

[theora_sample.bag.zip](https://github.com/foxglove/studio/files/6297714/theora_sample.bag.zip)

amacneil, 2021-04-12:
Awesome, thanks!

How this would work when connecting to a live node. Do you have any other nodes subscribing to this topic? Does it send the headers as soon as you subscribe or would we be picking it up mid stream?

rgov, 2021-04-12:
Good news, it does look like `theora_image_transport` sends the header to new subscribers. I suspect the first frames you receive might be inter-frames that are undecodable until the first intra-frame is sent.

https://github.com/ros-perception/image_transport_plugins/blob/noetic-devel/theora_image_transport/src/theora_publisher.cpp#L140-L146

The way the `image_transport` API works is that it broadcasts the same image once per installed transport. So `/camera` might be a raw, uncompressed `sensor_msgs/Image` message -- this is probably what most other nodes on the same computer would subscribe to for doing, say, object tracking.

But if you are streaming images to a node on another computer, or to a rosbag, you probably don't want to be saturating your network or logging those to disk or to the browser's memory. (With an HD camera, those uncompressed images can be 20+ MB *per frame*.) Thus the Theora, JPG, or PNG compressed transports are more useful.

rgov, 2021-04-12:
I put together an example of using ogv.js to decode raw Theora packets into video frames: https://github.com/rgov/js-theora-decoder ([Demo](https://rgov.github.io/js-theora-decoder/))

rgov, 2021-04-13:
Looks like the `b_o_s` field may be an easier way to test when a new video has started: https://github.com/ros-perception/image_transport_plugins/blob/noetic-devel/theora_image_transport/src/theora_subscriber.cpp#L129

rgov, 2021-05-24:
With @defunctzombie I have a working implementation of this in https://github.com/rgov/studio/tree/rzg/theora. There's also an earlier draft of this in #1067.

amacneil, 2021-12-26:
We do want to add support video decoding, but we will probably start with h.264 first (https://github.com/foxglove/studio/issues/2064). Let’s track in that ticket, and after we land it can consider other encodings.

Aposhian, 2022-05-19:
If @rgov has a working implementation of this, it seems like Theora might be a good first step towards supporting true video codecs. H.264 encoding is more widely used, but unlike theora, it has licensing issues, and does not have common ROS messages defined for it yet.

amacneil, 2022-05-20:
Video rendering is still on our radar (🥁😆) and we know we need it, but it's not top priority. If someone wants to contribute this, we're definitely open to it.

rgov, 2022-05-20:
Aside from coming up with a complete, multi-codec solution to video rendering, there are two immediate paths forward for the Theora support:

1. Spin it off into an extension. In this case, would someone from the team kindly look over it and tell me what API changes I might face in doing so? This code was written before the extension API was available.

    As an extension it would be able to render frames but it would not support annotations, the zoom settings, etc. that the regular Image panel has.

2. Finish the work that @esthersweon started to integrate this into the Image panel. (Maybe the branch was deleted?) In this case, if the team could help by stubbing it out again, I could fill in the implementation.

    Obviously decoding Theora is way more complicated than decoding other image formats due to all of the state you need to track for the codec. So it might not slide into the existing panel implementation trivially.

    On the other hand, there are literally only 4 image transport types in ROS (uncompressed, JPEG, PNG, and Theora) so it's not like the Image panel is going to keep growing new formats after this one until the ROS ecosystem as a whole figures out how to deal with other video codecs.