foxglove/studio #551: Feature request: Two y-axis scales, one plot panel

rgov, 2021-04-12:
I would like to plot values from two topics on the same panel, with the same *x*-axis scale but separate *y*-axis scales.

It is possible with Chart.js: https://www.chartjs.org/docs/latest/samples/line/multi-axis.html (Note this isn't a great example because you cannot tell which y-axis is used for which dataset.)

Effectively you would need to be configure zoom extents for both the left and right *y* axes, and then associate each topic with one axis or the other.

(from https://github.com/cruise-automation/webviz/issues/548)

amacneil, 2021-04-12:
Have you looked at the x-axis sync we added in v0.4? I think this would solve some use cases, since it makes it much easier to compare charts with different y-axes.

I could still see value in allowing multiple y-axes for each plot, but just curious whether you have tried this feature.

rgov, 2022-04-10:
This is still desirable to me as a way to increase the information density of my layouts.

rgov, 2022-04-25:
Relatedly, it looks like there is also now support to stack two plots that share an *x* axis: https://www.chartjs.org/docs/latest/samples/scales/stacked.html

defunctzombie, 2022-04-26:
That stacked plots feature is interesting!