foxglove/studio #605: Is there any way to publish the pose msg by click the point in 3D panel, just like the "2D Pose Estimate" button in rviz

iceuw, 2021-04-16:


jtbandes, 2021-04-21:
Not currently, but thanks for the suggestion! We can leave this issue open to track it.

jtbandes, 2021-08-02:
Another similar request (specifically for nav goals): https://github.com/cruise-automation/webviz/issues/664

amacneil, 2022-01-10:
Hi! So that we can prioritize this, it would be helpful if people could chime in with which of the following rviz features would be most helpful:

- 2D Pose Estimate (publishes `geometry_msgs/PoseWithCovarianceStamped` on `/initialpose`)
- 2D Nav Goal (publishes `geometry_msgs/PoseStamped` on `/move_base_simple/goal`)
- Publish Point (publishes `geometry_msgs/PointStamped` on `/clicked_point`)

Also: do you want to customize the topics that these messages are published on, or are you happy with the rviz defaults?

rowandempster, 2022-01-11:
@amacneil Thanks for taking this up 

For my immediate use case, 2D Nav Goal (publishes geometry_msgs/PoseStamped on /move_base_simple/goal) will suffice. Publish point functionality is also useful for sensor calibrations, and I would say is 2nd highest priority. These features can be released incrementally, i.e. my group would immediately benefit from 2D Nav Goal setting without the Publish Point functionality

dthakur, 2022-01-11:
2D Pose Estimate (publishes `geometry_msgs/PoseWithCovarianceStamped` on `/initialpose`) is most important for us. No changes needed from the current `rviz` behavior. Just need x, y and yaw published. This helps us with relocalization.

ccpowers, 2022-01-12:
@amacneil For our use case, we use both 2d nav goal (`geometry_msgs/PoseStamped` on a topic we call `/request/goal`) and 2d pose estimate (`geometry_msgs/PoseWithCovarianceStamped` on `/request/initial_pose`). We use them in simulation/development mostly, so it's easy enough to put a shim in to change topics if needed, especially since ours are not the standard topic names (but would be extra nice if we didn't have to). 

In the longer term, we have some other extensions that use interactive markers doing things like adding simulated objects and people to the environment which would be nice to port to foxglove.

amacneil, 2022-01-12:
Thanks everyone! This is helpful. We will take a look at it in the coming weeks.

schmittlema, 2022-01-14:
I would love all 3 with the default topics but priority would be.
1. 2D pose estimate
2. 2D nav goal
3. 2D point

jtbandes, 2022-01-14:
Today we learned rviz has a Tool Properties panel where you can set the covariance that's used in the 2D Pose Estimate. [image]

jtbandes, 2022-01-26:
RViz's pose/goal publishing shows an arrow and fixes the length of the arrow, rather than allowing 2 arbitrarily located points:

https://user-images.githubusercontent.com/14237/151252579-5a8b6057-4dc9-48f1-8eac-6cdc62e39d7a.mov

foxymiles, 2022-01-26:
Enhancements:

1. Use arrow of fixed length for pose drawing.
2. Add link to settings from the popout menu.
3. Use message type in popout menu instead of `goal`, `pose`, `point`
4. Improve icons?

foxymiles, 2022-01-27:
> RViz's pose/goal publishing shows an arrow and fixes the length of the arrow, rather than allowing 2 arbitrarily located points:
> 
>  Screen.Recording.2022-01-26.at.1.46.50.PM.mov

Thanks for the reference. I've updated it to work more like this now.

PR #2719 (2022-01-28): Implement click to publish points & poses in 3d panel.
**User-Facing Changes**
This adds a feature that allows users to click on the 3d panel to publish various point & pose messages.

**Description**
This slightly the existing state logic in the 3d panel toolbar to consolidate it into a shared reducer and adds new buttons and tools for publishing points & poses. It also adds new 3d panel settings to customize how these messages are formed.

Fixes #605
closed 2022-01-28 by foxymiles