foxglove/studio #678: Loading the production version (0.5 and 0.6 releases) takes several seconds

amacneil, 2021-04-27:
This wasn't happening in earlier versions. @defunctzombie thinks it is due to loading all the panel code up front now.

defunctzombie, 2021-04-28:
This commit helped some by breaking up some of the larger script files which allowed us to turn on minification. https://github.com/foxglove/studio/commit/fed04ec37c6b0bb88153eb52ad56b4331846ca84

The next change would be to avoid loading the panel code altogether. I've created https://github.com/foxglove/studio/issues/747 to track that.

amacneil, 2021-04-28:
This is still a problem with prod builds on latest main. It takes about 6 seconds from when the app is displayed until when the app is responsive. Until that time, it's not even possible to drag the application window.

amacneil, 2021-04-28:
I tested some older versions:

- v0.4: startup freeze for 1-3 seconds
- v0.5 and above: startup freeze for 6-7 seconds

amacneil, 2021-05-12:
Fixed by https://github.com/foxglove/studio/pull/863

closed 2021-05-12 by amacneil