foxglove/studio #681: WebGL error when coloring point cloud by point data

amacneil, 2021-04-27:
Sample bag: https://storage.googleapis.com/cartographer-public-data/bags/backpack_3d/with_intensities/b3-2016-04-05-14-14-00.bag

Repro by turning on `horizontal_laser_3d` topic and coloring by intensity.

[image]

This bag also has issues coloring `/scan_cloud_aggregated` by z, although x and y work: 
https://storage.googleapis.com/cartographer-public-data/bags/taurob_tracker/taurob_tracker_simulation.bag

via @jtbandes

amacneil, 2021-07-19:
Can't repro in latest version, both bags color fine for me.

[image]

The first bag does seem to cause our app to crash on my machine, I will open a separate ticket.

jtbandes, 2021-07-19:
Did you check the console to see whether there are WebGL errors (rather than just looking at 3D view visually)?

amacneil, 2021-07-19:
Yes. No console errors for me on either bag following the instructions above. Maybe good if you test as well to ensure it's not a platform specific thing?

jtbandes, 2021-07-19:
I still see the error on mac. Layout: 

```
{
  "configById": {
    "3D Panel!1avl4ju": {
      "checkedKeys": [
        "name:Topics",
        "t:horizontal_laser_3d"
      ],
      "expandedKeys": [
        "name:Topics"
      ],
      "followOrientation": false,
      "cameraState": {
        "distance": 75,
        "perspective": true,
        "phi": 0.7853981633974483,
        "target": [
          0,
          0,
          0
        ],
        "targetOffset": [
          0,
          0,
          0
        ],
        "targetOrientation": [
          0,
          0,
          0,
          1
        ],
        "thetaOffset": 0,
        "fovy": 0.7853981633974483,
        "near": 0.01,
        "far": 5000
      },
      "modifiedNamespaceTopics": [],
      "pinTopics": false,
      "settingsByKey": {
        "t:horizontal_laser_3d": {
          "colorMode": {
            "mode": "rainbow",
            "colorField": "intensity"
          }
        }
      },
      "autoSyncCameraState": false,
      "autoTextBackgroundColor": true,
      "diffModeEnabled": true
    }
  },
  "globalVariables": {},
  "userNodes": {
    "7662ada8-06c0-4969-a77d-64a151101dd8": {
      "sourceCode": "import { Input, Messages } from \"ros\";\n\ntype foo_msgs__Bar = {\n  a: string;\n  b: number;\n  // z?: 3;\n  // (x: 3): foo_msgs__Bar;\n  // @ts-ignore\n  // qza;\n};\ntype GlobalVariables = { id: number };\n\nexport const inputs = [\"/radar/tracks\"];\nexport const output = \"/studio_node/test\";\n\n// Populate 'Input' with a parameter to properly type your inputs, e.g. 'Input<\"/your_input_topic\">'\nconst publisher = (\n  message: Input<\"/radar/tracks\">,\n  globalVars: GlobalVariables\n): foo_msgs__Bar => {\n  return {\n    a: \"12\",\n    b: 2,\n  } as foo_msgs__Bar;\n};\n\nexport default publisher;\n",
      "name": "/studio_node/7662ada8"
    }
  },
  "linkedGlobalVariables": [],
  "playbackConfig": {
    "speed": 1,
    "messageOrder": "receiveTime",
    "timeDisplayMethod": "ROS"
  },
  "layout": "3D Panel!1avl4ju"
}
```

amacneil, 2021-07-20:
No error for me with 0.13.1 on linux - interesting. Seems odd that it would be platform dependent.

Will assign to you since you can repro.

jtbandes, 2021-07-20:
Not the first platform dependent WebGL bug :trollface:  https://github.com/cruise-automation/webviz/pull/24

PR #1559 (2021-07-24): Fix PointClouds out of bounds read for non-rgb color field
**User-Facing Changes**
Fixed a bug where point clouds would fail to render with certain color settings selected.

**Description**
Fixes #681

The PointClouds rendering command used the same shader for rendering a single-valued colorField and a multi-valued (`rgb`) colorField. The shader was hard-coded to always accept `vec3 color`, but only `color.x` was actually used for single-color modes. When a single-color mode was used and the colorField occurred near the end of the point struct, this caused an out-of-bounds read.

[image]

(In  [... cut]
commit 9b9f9504 (2021-07-24): Fix PointClouds out of bounds read for non-rgb color field (#1559)

**User-Facing Changes**
Fixed a bug where point clouds would fail to render with certain color settings selected.

**Description**
Fixes #681

The PointClouds rendering command used the same shader for rendering a single-valued colorField and a multi-valued (`rgb`) colorField. The shader was hard-coded to always accept `vec3 color`, but only `color.x` was actually used for single-color modes. When a single-color mode was used and the colorField occurred near the end of the point struct, this caused an out-of-bounds read.

To f [... cut]
closed 2021-07-24 by jtbandes