foxglove/studio #686: Node Playground doesn't support dynamically updating datatypes

amacneil, 2021-04-27:
via @jhurliman:

Ros1Player dynamically updates datatypes when connection headers are received from publishers, but Node Playground doesn't support dynamic updates of datatypes. This leads to nodes not compiling because the datatype TS definitions ("rosLib") are empty.

Start the `ros1-turtlesim-test` container, connect with FGS, and attempt to compile the following script in Node Playground:

```ts
import { Input, Messages } from "ros";

type GlobalVariables = { id: number };

export const inputs = ["/turtle1/color_sensor"];
export const output = "/webviz_node/color_sensor";

// Populate 'Input' with a parameter to properly type your inputs, e.g. 'Input<"/your_input_topic">'
const publisher = (
  message: Input<"/turtle1/color_sensor">,
  globalVars: GlobalVariables
): Messages.turtlesim__Color => {
  return { ...message };
};

export default publisher;
```

It fails with:

```
Cannot use namespace 'Input' as a type.(2709)
Namespace '"ros"' has no exported member 'turtlesim__Color'.(2694)
```

The error happens when you don't have an input source loaded. Maybe in that case we should produce a different more explicit error.

Once you connect an input source the input error goes away. Other errors might also go away depending on what messages your input source has.

defunctzombie, 2021-06-24:
This issue says it happens when `... you don't have an input source loaded` but earlier it says to connect to the container which would load an input source. I'm not following what the reproduction case here is. Is this a bug or sub-par user experience? i.e. once you are connected you do get the datatypes? My experience has been that is the case.

amacneil, 2021-11-19:
Fixed by https://github.com/foxglove/studio/pull/586.

If anyone comes across a similar error, please open a new issue.

closed 2021-11-19 by amacneil