foxglove/studio #693: Reset all panel state when /run_id changes

amacneil, 2021-04-27:
via @jhurliman:

We can detect when the robot has been torn down and restarted (vs a network disconnect/reconnect) by monitoring the `/run_id` parameter in Ros1Player. When this changes, all panel state should be cleared so the user doesn't see a mixture of old and new session state in panels like the watchdog monitor, timeline chart, 3D view markers, etc.

wkalt, 2021-05-18:
@amacneil @jhurliman I think a user can easily hit this in our onboarding flow. They will initially load the welcome screen with the demo bag data, and then the first time they try to use their own bag they will see mixed state, unless the new set of topics matches what was previously displayed.

amacneil, 2021-05-18:
agreed, let's prioritize this

jtbandes, 2021-05-18:
That doesn't sound right - loading their own bag will be swapping out a data source, since the demo bag is a separate HTTP bag. This issue only affects you if you're using a ROS/WebSocket connection and tearing down/launching nodes without restarting the ros master.

amacneil, 2021-06-10:
Related: reset panel state when player changes https://github.com/foxglove/studio/pull/1008

amacneil, 2021-12-08:
@jhurliman is this still relevant?

jhurliman, 2021-12-15:
Yes it is