foxglove/studio #699: Convert Modals to FluentUI

amacneil, 2021-04-27:
via @jtbandes:

- [ ] Convert HelpModal to FluentUI
- [ ] Convert NoHeaderTopicsButton modal to FluentUI
- [ ] Convert NotificationDisplay modal to FluentUI
- [ ] Convert ShareJsonModal to FluentUI
- [ ] Convert useConfirm modal to FluentUI
- [ ] Convert 3d panel topic settings modal to FluentUI
- [ ] Convert ColorPickerOverlay modal to FluentUI

2metres, 2021-07-06:
Removing non modals and creating separate issues

2metres, 2021-07-06:
Once these are done should we remove the generic foxglove `<Modal /> ` and `<RenderComponentToBody />`?

jtbandes, 2021-07-06:
If that's the only thing RenderComponentToBody is used for, then yep

defunctzombie, 2021-07-21:
@2metres All the items in this list are done. Are we good to close this? Did the foxglove `<Modal />` and `<RenderComponentToBody/ >` get removed? Should we make a separate issue for that?

PR #2128 (2021-11-11): Remove RenderToBodyComponent
**User-Facing Changes**
None

**Description**
Removes RenderToBodyComponent

Closes #699
closed 2021-11-11 by defunctzombie