foxglove/studio #707: Better support for topic latching when scrubbing

amacneil, 2021-04-27:
When scrubbing, we should intelligently find the right message to display for topics which are published infrequently.

References:
- https://github.com/cruise-automation/webviz/issues/314
- https://github.com/fetchrobotics/webviz/commit/ab301ca317e816583db6e93d24b370253c5c2092

rgov, 2021-05-21:
I'm not sure what you have in mind but it would be helpful to have a `PanelAPI` function that enumerates messages forward or backward in a given time window.

This be useful for things like video streams (#537), where to decode a given frame you need to find some earlier messages that provide context: the header(s) that describe the width, height, and other parameters; the last intra frame; and all subsequent inter frames that are deltas to the inter frame.

1. Search backwards from `currentTime` to find the header packet message.
2. Search backwards from `currentTime` to `headerPacket.header.stamp` to find the most recent intra frame.
3. Search forwards from `intraFrame.header.stamp` to `currentTime` to collect a list of inter frames.

jhurliman, 2022-01-08:
This may be related to <https://github.com/foxglove/studio/issues/2343>. Internally latching every topic in Studio would likely make seeking/scrubbing work a lot better in practice, especially with bags that have `/tf_static`.

amacneil, 2022-02-23:
Also created https://github.com/foxglove/studio/issues/2899 for the same feature in Data Platform

defunctzombie, 2022-03-11:
Re-opening because this issue is a meta-tracking issue which also needs to have support for mcap.

defunctzombie, 2022-06-13:
@rgov 

> I'm not sure what you have in mind but it would be helpful to have a `PanelAPI` function that enumerates messages forward or backward in a given time window.

This issue won't cover this functionality. This issue is scoped to only finding the last message on each topic.

PR #2915 (2022-03-11): Bag playback with forward and reverse iterators
**User-Facing Changes**
Add a new feature flag for "Experimental bag player".

When enabled, using a bag data source (remote or local) will use a new player with new seeking capabilities. Seeking to any location will load the last message on each subscribed topic into panels. This lets panels display the last message on a topic even if it occurred much earlier in the file.

**Description**
This change introduces a new IterablePlayer which uses iterators for message reading and preloading. When seeking, the player uses a reverse iterator to backfi [... cut]
commit b470dbd3 (2022-03-11): Bag playback with forward and reverse iterators (#2915)

**User-Facing Changes**
Add a new feature flag for "Experimental bag player".

When enabled, using a bag data source (remote or local) will use a new player with new seeking capabilities. Seeking to any location will load the last message on each subscribed topic into panels. This lets panels display the last message on a topic even if it occurred much earlier in the file.

**Description**
This change introduces a new IterablePlayer which uses iterators for message reading and preloading. When seeking, the player uses a reverse iterator  [... cut]
closed 2022-06-14 by defunctzombie