foxglove/studio #708: Panel disappears if dragged back and forth between Tab mosaics

amacneil, 2021-04-27:
via @jtbandes:

I can consistently repro this in the storybook scenario I newly added in https://github.com/foxglove/studio/pull/410.

This bug also exists in upstream webviz when tested with the same layout.

https://user-images.githubusercontent.com/637671/116322875-c862e700-a771-11eb-8a9d-f29f914489e0.mov

jtbandes, 2021-05-03:
There is another bug, probably related, which causes an error when a drag from a tab is **canceled** (press esc while dragging):

https://user-images.githubusercontent.com/14237/116940186-48bd9800-ac09-11eb-84a9-8177b672c62f.mov

[Mosaic drag testing2.json.zip](https://github.com/foxglove/studio/files/6418086/Mosaic.drag.testing2.json.zip)

defunctzombie, 2021-10-22:
@amacneil I'm no longer able to reproduce this with 0.20.1 (with the production app). Maybe the ticket @jtbandes referenced fixed this or other fixes we've made to id tracking. Going to close this as neither issue shown in the ticket I am able to reproduce. If you can reproduce with the app or current stories please re-open with new instructions.

foxymiles, 2021-11-10:
@jtbandes I think I have a fix for the crash adding a new panel. I'm looking into the panel disappearance issue now.

foxymiles, 2021-11-10:
I found an easier repro using the layout @jtbandes posted. If you drag the panel out of its original container, then try to drag it back but position the cursor so that no drop target is indicated and then drop the panel it just vanishes when you complete the drop.

https://user-images.githubusercontent.com/93935560/141116493-6e7640da-2e5d-43e0-9311-6a06d8c0b405.mp4

foxymiles, 2021-11-10:
https://github.com/foxglove/studio/pull/2132 Should fix the crash on adding a  sub panel to a fresh tab panel.

amacneil, 2021-11-19:
Noting here that we have not seen any issues here that don't involve the tab panel.

From @foxymiles: This issue is possibly related to garbage collection that runs at the end of the tab layout change process.

jtbandes, 2021-11-19:
The root cause of https://github.com/foxglove/studio/issues/708#issuecomment-965103972 is that `sourceTabChildConfigs` doesn’t account for when the target tab is itself a child of the source tab, so the source child config ends up overwriting the updated target tab config with its original value here:

[image]

closed 2021-11-20 by jtbandes