foxglove/studio #710: Ctrl+~ controls are not clickable (MacOS)

amacneil, 2021-04-27:
via @jhurliman:

Holding `~` brings up the secret UI for managing panels, but if you hold `Ctrl+~` the secret UI still appears but none of the buttons are clickable and dragging no longer works. We should either disable the menu from appearing with this combo or allow the buttons to work.

For some reason it works fine when holding `Cmd+~` or `Option+~`.

jtbandes, 2021-04-28:
I would say this is behaving as expected because ctrl+click on macOS is the same as right-click. That said, I also think the use of a printable character `` ` `` as the modifier key is a bit strange — we might want to think about using something like alt/option instead.

amacneil, 2021-04-28:
Agreed. However, if using it with Ctrl is unsupported we should hide "rearrange mode" as soon as you start holding Ctrl.

Can you think of any prior art for which modifier key would be appropriate to throw the app into rearrange mode?

It might also need to differ between platforms. Option is maybe unused enough on macOS, but we definitely wouldn't want to override Ctrl or Alt on Windows.

Maybe keeping `~` is fine but we disable using it with any modifiers?

amacneil, 2021-07-26:
Closing as won't fix - we want to address panel management shortcuts in the future, but this particular annoyance is a macOS-specific issue due to Ctrl-click mapping to right-click.

closed 2021-07-26 by amacneil