foxglove/studio #712: Switch to comlink instead of our home grown rpc

amacneil, 2021-04-27:
via @defunctzombie:

https://github.com/GoogleChromeLabs/comlink#typescript

Supports better type information for the receiver. And one less thing to maintain with strange edge cases.

amacneil, 2021-07-07:
@defunctzombie I imported this but don't know what it is referring to. should we close it or do you want to flesh out the description?

jtbandes, 2021-07-07:
We have https://github.com/foxglove/studio/blob/main/packages/studio-base/src/util/Rpc.ts which we use to communicate with workers. Comlink does a lot of the same things and claims to have TypeScript support so we should explore using it to replace our custom implementation.

amacneil, 2021-07-07:
Oh ok, for web workers. I thought this was related to the electron RPC or something.