foxglove/studio #713: Parameter Panel: Improve display of long or binary parameter values

amacneil, 2021-04-27:
via @jhurliman:

We can use the raw message panel as a reference. UInt8Array values can be rendered as `(${length}) [${first20Values}, …]` while long string values can be truncated with ellipses.

In both cases it would be nice to have a save to disk button.

foxymiles, 2022-01-14:
@amacneil @jhurliman What would we save to disk? The binary data of a single parameter or all params in JSON or something like that?

amacneil, 2022-01-14:
I imported this comment from our old ticket system, so I think it needs to either be expanded with more specifics or closed.

jhurliman, 2022-01-14:
Save to disk really should be broken out into a separate ticket, but I would advocate saving all params to JSON and base64 encoding Uint8Array fields.