foxglove/studio #715: ROS 2 native connection support

amacneil, 2021-04-27:


shoemakerlevy9, 2021-07-03:
Hi, my company Rover Robotics is launching a robot in September via Kickstarter that is going to ship with ROS2 Foxy by default. We are looking for an alternative to RVIZ because we don't like QT development. I really like the look of Foxglove and would like to try it out. Is there a work a round for getting it to work with ROS Foxy?

amacneil, 2021-07-05:
It is apparently possible to use Foxglove Studio with ROS 2 via the [ros1 bridge](https://github.com/ros2/ros1_bridge) (I haven't tried this myself but heard from someone doing it). If you try this we would love to hear back on how it works out for you.

Support for ROS 2 without needing the bridge is being actively worked on (e.g. [1](https://github.com/foxglove/rosmsg2-serialization) [2](https://github.com/foxglove/rosbag2) [3](https://github.com/foxglove/cdr)). I can't give a concrete timeline for when it will be complete, but it should be ready :soon:.

ShawnSchaerer, 2021-08-09:
We use it with ROS1 bridge.  Works fine but it's an added step.

amacneil, 2021-08-10:
Update on this ticket status:

- ROS 2 bag playback (https://github.com/foxglove/studio/issues/717) is launched
- ROS 2 websocket (rosbridge) connection support is in progress (see https://github.com/foxglove/studio/issues/716 - we are in the process of publishing a new version of rosbridge, and there are some minor issues to fix in studio)
- ROS 2 native connection support is in progress, but will take a while (~months) since it requires reimplementing all of ROS 2 + DDS in JS.  You can follow this development progress in our various [ROS 2 typescript libraries](https://github.com/foxglove?q=ros2&type=&language=&sort=). In the meantime rosbridge will be our recommended path for ROS 2 users.

jhurliman, 2021-08-14:
ROS 2 native connection progress can be tracked on the pull request at https://github.com/foxglove/studio/pull/1640

jtbandes, 2021-09-03:
Gonna keep this issue open since it seems like the connection is not yet in a usable state (#1750)

defunctzombie, 2021-09-07:
We'll close this as the overall ROS2 Native support is in place. We have the new issues to track smaller features/bugs.

closed 2021-09-07 by defunctzombie