foxglove/studio #716: ROS 2 websocket connection support

amacneil, 2021-04-27:
Placeholder ticket for anyone to discuss if this is interesting to them.

In theory we could allow connections over websocket (similar to what we support with `rosbridge` in ROS 1) using https://github.com/RobotWebTools/ros2-web-bridge.

However, we have also discussed a native-connection-over-ssh option which might be more useful than websocket connections. We're open to feedback.

See also:
- https://github.com/foxglove/studio/issues/715

amacneil, 2021-04-28:
This might not be necessary if we add support for ssh tunnels.

However, it may be needed if we wish to retain browser compatibility.

jtbandes, 2021-08-02:
Leaving this open until https://github.com/RobotWebTools/rosbridge_suite/pull/574 is merged and released.

jtbandes, 2021-08-11:
Release is currently blocked on https://github.com/GT-RAIL/rosauth/issues/35

jtbandes, 2021-08-13:
- galactic: https://github.com/ros/rosdistro/pull/30479
- foxy: https://github.com/ros/rosdistro/pull/30480

amacneil, 2021-08-23:
All studio features are complete, we are just waiting for the next Galactic sync to release rosbridge, which is happening on [Tuesday](https://discourse.ros.org/t/preparing-for-galactic-sync-2021-08-24/21900).

PR #1480 (2021-07-28): Add ROS 2 support to RosbridgePlayer
**User-Facing Changes**

**Description**
Fixes #716
Depends on https://github.com/RobotWebTools/rosbridge_suite/pull/574
closed 2021-08-23 by amacneil