foxglove/studio #723: Interactions for user-drawn polygons in 3D panel are broken

esthersweon, 2021-04-28:
[image]

Of the functionality described above, only 1) drawing a polygon, 2) selecting a polygon, and 3) bisecting a polygon edge seem to be working. Selecting a point within the polygon, moving / resizing / deleting a polygon, deleting a point in the polygon, etc. are not working, at least not as described by the included documentation.

amacneil, 2021-07-26:
@esthersweon is following up to get more context on this feature

amacneil, 2021-07-26:
Likely that we will replace this feature with ability to create "static markers" in the scene (https://github.com/foxglove/studio/issues/1560).

In the meantime, we should update docs to remove references broken functionality, since it's not clear anyone is using this feature.

esthersweon, 2021-07-26:
Used mostly for debugging - people copy these points from someplace else (e.g. test outputs) and paste them into the 3D panel to see where it is related to other objects in the scene.

amacneil, 2021-07-26:
Next steps: Move entire polygon feature behind feature flag and remove documentation.

PR #1562 (2021-07-27): Hide the 3D panel's polygon drawing behind a feature flag
**User-Facing Changes**
The sidebar controls for drawing polygons in the 3D panel are hidden, unless the feature is explicitly turned on in the app settings.

**Description**
Hides polygon drawing feature behind feature flag; removes documentation. Will essentially sunset this feature, barring any active use cases that come up during this process.

Resolves #723.
closed 2021-07-27 by esthersweon