foxglove/studio #726: Node Playground should support customizing datatype name

amacneil, 2021-04-28:
via @jtbandes:

Node Playground lets you publish custom datatypes, but the datatype name that's exposed to the rest of the app is autogenerated. This means the message can't be used for other panels that are looking for specific datatypes.

There is currently a hack specifically to allow publishing `visualization_msgs/MarkerArray`: 

https://github.com/foxglove/studio/blob/860b4df2cad85a39bca022749f80a0a48c6a8796/app/players/UserNodePlayer/nodeTransformerWorker/typescript/ast.ts#L322-L323

It would also be good to provide built-in definitions for the common ROS datatypes.

[image]

jhurliman, 2021-05-20:
> It would also be good to provide built-in definitions for the common ROS datatypes.

This part is addressed in https://github.com/foxglove/studio/pull/1038

jtbandes, 2021-06-17:
> > It would also be good to provide built-in definitions for the common ROS datatypes.
> 
> This part is addressed in #1038

I don't think it is — I tried to use Node Playground today and I didn't find any common data types exposed to my code. I filed #1261 to track this part

defunctzombie, 2022-02-18:
Node playground understands if you are using existing messages from your datasource types (and builtin types) and uses those datatype names.

closed 2022-02-18 by defunctzombie