foxglove/studio #729: Improve Image panel zooming behavior

amacneil, 2021-04-28:
via @jtbandes:

- Zooming with scroll wheel / trackpad pinch feels too slow
- Zoom percentage / menu looks strange – the percentage itself may not be that useful, and these menu items could probably be buttons instead for better UX
- Easy to get in a weird state where the image doesn't stay aligned as you resize the panel

mwisniowski, 2021-04-29:
Requesting to increase the zoom limit to 5000%

For camera applications there are times when you want to zoom in to a pixel level, eg when manually focussing a lens

jhurliman, 2021-05-20:
@mwisniowski https://github.com/foxglove/studio/pull/1046 increases the default max zoom to 300% and makes it a panel setting, so you can turn it up to 5000%. It also switches the default image scaling from bilinear to nearest neighbor, although this can be toggled in the image panel settings as well.

closed 2021-05-20 by jhurliman