foxglove/studio #741: Enable React Strict Mode for all panels

jtbandes, 2021-04-28:
Explanation from #627:
> Read more about strict mode at https://reactjs.org/docs/strict-mode.html
> 
> > StrictMode is a tool for highlighting potential problems in an application. …  It activates additional checks and warnings for its descendants.
> 
> > Strict mode checks are run in development mode only; they do not impact the production build.
> 
> The most notable behavior of strict mode is that it causes component render functions, useMemo callbacks, and other kinds of functions that should be stateless to be called **twice per render** in dev mode. This may cause problems in components that use these in side-effectful ways, so I didn't want to immediately enable it by default in all panels. I added a warning icon in the panel toolbar to prompt us to enable it in existing panels incrementally:
> 
> [image]

The panels below all have `supportsStrictMode` set to `false`. To complete one of these tasks, first remove the false value, then manually test the panel features to ensure strict mode doesn't cause any regressions. For an example of how to recognize and fix problems caused by strict mode, see https://github.com/foxglove/studio/pull/627#discussion_r618131836.

- [ ] Internals
- [ ] NumberOfRenders
- [ ] SubscribeToList
- [ ] Diagnostic Detail
- [ ] Diagnostic Summary
- [ ] Global Variables
- [ ] Global Variable Slider
- [ ] Image View
- [ ] Internal Logs
- [ ] Map
- [ ] Node Playground
- [ ] Parameters
- [ ] Playback Performance
- [ ] Plot
- [ ] Publish
- [ ] Raw Messages
- [ ] Source Info
- [ ] State Transitions
- [ ] Tab
- [ ] Table
- [ ] ThreeDimensionalViz
- [ ] Welcome

Finally, once all panels support strict mode:
- [ ] Delete the `supportsStrictMode` setting and always render panels in StrictMode.

jtbandes, 2022-01-08:
State Transitions and Plot blocked on: https://github.com/facebook/react/issues/23082

closed 2022-01-12 by jtbandes

amacneil, 2022-01-12:
🎉