foxglove/studio #747: Lazy load panel components on demand

defunctzombie, 2021-04-28:
Rather than importing all panels into the BuiltinPanelCatalog - make a catalog from instances (maybe of a Panel class) which will lazy load panel components on demand. This will lower the amount of code loading up-front to render the initial app containers.

jtbandes, 2021-04-28:
I think a nice loading screen would be more impactful for “smooth launch experience” than splitting panel loading. _Now you have 10 loading spinners_

amacneil, 2021-04-28:
@jhurliman made a good comment in Slack about following the pattern of showing some greyed out app UI while loading (similar to pretty much every other tool & website these days). 

The main problem in v0.6 is that the whole window is unresponsive (not even draggable) while we're loading. I believe Roman fixed that in https://github.com/foxglove/studio/pull/639.

PR #1474 (2021-07-19): Internal: Lazy load builtin panels
**User-Facing Changes**
None.

**Description**
Rather than import all builtin panels, use dynamic imports to only
load the builtin panel module when added to the layout.

Using dynamic loading for panels decreases the size of the main bundle which helps
with initial load times.

Fixes #747
closed 2021-07-19 by defunctzombie