foxglove/studio #751: Some 3d panel stories are broken

amacneil, 2021-04-28:
via @jtbandes:

Many 3D panel stories appear to be blank, but the story title indicates that some object should be selected. This also seems to be a problem in the webviz repo, but presumably at some point in history there was an object visible here and not just a black screen. There are also multiple errors getting thrown because of confusion between `onMount` and `onFirstMount` in the PanelSetup helper. Likely rewriting PanelSetup to use hooks and cleaning up these handlers—shouldn't everything be `onFirstMount`?—would help resolve these problems

[image]

defunctzombie, 2022-02-18:
I can't find these stories anymore.

closed 2022-02-18 by defunctzombie

jtbandes, 2022-02-18:
Looks like they were deleted in https://github.com/foxglove/studio/pull/2403. I think this might have been a mistake since the commit message mentions deleting stories about drawing tools, but only 1 of the 4 deleted stories mentioned drawing tools.