foxglove/studio #753: Remove `from "@foxglove-studio/app/stories/"` in panels and non-stories

amacneil, 2021-04-28:
via @defunctzombie:

Importing the story specific utils or whatever is in that folder should not happen in regular components. If we need a platform interface for stories we should hoist that our of stories. Working on stories should be low barrier and low impact risk to production.

via @jtbandes:

Almost all of these are just the `inScreenshotTests` function. Some "non-story" files that import these are just story support code that aren't themselves stories. Some others are mock-related. So I think the only todo is perhaps a minor organization/naming cleanup

jtbandes, 2021-07-08:
@defunctzombie Do we see any value in keeping this issue open? Is the task now to remove `inScreenshotTests` or is it not worth the effort?

defunctzombie, 2021-07-08:
Yep - removing `inScreenshotTests` is would be the next step. Here are the remaining files that reference inScreenshotTests which are not tests or stories themselves.

```
packages/studio-base/src/panels/ImageView/index.tsx:import inScreenshotTests from "@foxglove/studio-base/stories/inScreenshotTests";
packages/studio-base/src/panels/NodePlayground/Editor.tsx:import inScreenshotTests from "@foxglove/studio-base/stories/inScreenshotTests";
packages/studio-base/src/panels/ThreeDimensionalViz/TopicTree/Layout.tsx:import inScreenshotTests from "@foxglove/studio-base/stories/inScreenshotTests";
packages/studio-base/src/panels/ThreeDimensionalViz/World.tsx:import inScreenshotTests from "@foxglove/studio-base/stories/inScreenshotTests";
```

I don't think components should know they are under test. They can have properties that turn certain things on or off.

amacneil, 2021-11-19:
@defunctzombie says "close it"

closed 2021-11-19 by amacneil