foxglove/studio #773: Ability to export mp4 files

amacneil, 2021-04-28:
It would be great if it were possible to export the current layout+data source as an mp4.

For example, if a bag is loaded, and you configure the panel, you could hit `File -> Export video` and we would render a smooth video at a desired resolution/framerate (rather than needing to use screen recording tech which would not result in a smooth framerate).

See also: Ability to export single frames as pngs (https://github.com/foxglove/studio/issues/772).

amacneil, 2021-04-28:
via @jhurliman:

I think this is how you do framerate-independent video capture in an electron app:

- use https://www.electronjs.org/docs/api/desktop-capturer to stream the electron renderer window to an offscreen `<canvas>` as a video feed. example glue code at https://ourcodeworld.com/articles/read/280/creating-screenshots-of-your-app-or-the-screen-in-electron-framework
- Create a `MediaRecorder` feed from a `<canvas>` (see https://webrtc.github.io/samples/src/content/capture/canvas-record/)
- Use the magical `CanvasCaptureMediaStreamTrack.requestFrame()` method to feed frames one at a time into the `MediaRecorder` (see https://developer.mozilla.org/en-US/docs/Web/API/CanvasCaptureMediaStreamTrack/requestFrame)
- Run a loop that steps rendering forward one tick and calls `requestFrame()` once until recording end

jtbandes, 2021-04-28:
Please note there is already a bunch of code around video recording:

https://github.com/foxglove/studio/blob/main/app/players/automatedRun/videoRecordingClient.ts

https://github.com/cruise-automation/webviz/blob/master/packages/webviz-core/shared/recordVideo.js

https://github.com/cruise-automation/webviz/blob/master/packages/webviz-core/script/record-local-bag-video.js

jhurliman, 2021-11-27:
I do!

samzapo, 2022-10-21:
This feature for exporting the output of the 3D pane to MP4 would be extremely useful.

Right now, we are using rather large meshes for visualization, so the visualizer can sometime lag behind real-time.

Being able to render the 3D viewer into a video that runs at the same rate as our log file's timestamp would make showing off our visualizations much easier and look more realistic and visually appealing.

jtbandes, 2022-10-21:
> Right now, we are using rather large meshes for visualization, so the visualizer can sometime lag behind real-time.

Are the meshes updating frequently? While it may take some time to ingest a mesh from a message, it should be relatively fast to update its pose. If there is a performance problem here it would be good for us to improve that behavior. It would help if you can share more details about the mesh, or ideally some sample data we can use to reproduce the lag.