foxglove/studio #777: Ability to load multiple sequential bag files

amacneil, 2021-04-28:
For example, if someone ran rosbag record with `--split`, they would end up with a bunch of bag files in a folder. It would be nice if we could load them all (or a subset) at once.

From public discussion in ROS discord:

[image]

jtbandes, 2021-04-28:
Also requested by @jackma.

luqnaim, 2022-01-03:
Bump. Noticed this is already possible with webviz where the second bag file topics would have a prefix of "webviz_source_2". Though, the limit for webviz is only 2 bag files at once. Would be cool if foxglove could extend this further to load N number of bag files!

amacneil, 2022-01-03:
The specific request in this PR is to be able to load multiple sequential files (e.g. if you split the bag files based on size).

Webviz does not support this, it only allows you to compare two bag files covering the same time range, with a prefix for the second bag file (e.g. if you re-simulated some period of time and wanted to compare outputs). We have that feature request tracked in a separate issue (https://github.com/foxglove/studio/issues/2120).

So @luqnaim are you looking for loading multiple sequential files, or do you want to compare two different files of the same time range (with a topic prefix)?

Also note here that the workaround (for the original request in this PR) is to first use the rosbag CLI tool to merge/split/etc bag files as needed, before loading them into Foxglove Studio. You could also upload the bag files to Foxglove [Data Platform](https://foxglove.dev/data-platform), because when we play remote data we combine all topics over the requested time range, regardless of how many separate files you started with.

luqnaim, 2022-01-03:
@amacneil ah, thank you for the link. I did not come across that issue during my search. Answering your question, issue https://github.com/foxglove/studio/issues/2120 is exactly what i am looking for. I would like to inspect the behavior of multiple algorithms on the same video sequence and be able to compare them frame by frame. I was able to do this with webviz but was limited to only 2 simultaneous bag files.

valschmidt, 2022-03-06:
+1 for this one. 

We have a tool, [bag_manager](https://github.com/CCOMJHC/bag_manager), which allows one to select start/stop times and topics to include/exclude from a collection of bag files and writes a new bag file from the result to work around this. But it would be nice if Foxglove could handle this natively - point to a directory rather than a file, index the whole thing, then let you interactively play with everything there.

jtbandes, 2022-03-07:
We might want to also think about what happens if the bags are not totally sequential but overlapping (such as if there were two separate bag recorders both using `--split`, and splitting at different times), and if there is some duplicate message data between bags (such as if two recorders both happened to record one or a handful of small 'core' or 'metadata' topics that users intentionally wanted to be present in all bag files — or even if it happened by accident).

joegeisz, 2022-05-23:
I would also love to see this feature!