foxglove/studio #781: Support Ouster lidar packets

amacneil, 2021-04-28:
via @jtbandes:

We currently support Velodyne packets. We should add support for Ouster too.

See:
- https://github.com/cruise-automation/webviz/issues/162 (similar webviz feature request)
- https://data.ouster.io/downloads/software-user-manual/lidar-packet-format-update.pdf
- https://github.com/ouster-lidar/ouster_example

amacneil, 2021-11-19:
Closing because we have not had any user requests for this. If interested, please comment.

albait03, 2022-02-17:
Hello, 
i am currently working with Lidar from Ouster. When I am trying to visualize the data i get the following error:

[image]

Is there any plan to support visualizing bag files from Ouster in the near future?

amacneil, 2022-02-17:
See also: https://github.com/cruise-automation/webviz/issues/714

@albait03 I saw you posted here and then deleted your comment. Did you try your bag file in Foxglove Studio (https://studio.foxglove.dev/)? I'm pretty sure it should work out of the box based on your screenshot.

albait03, 2022-03-08:
Thank you for the advice. I tried out Foxglove Studio and it met my requirements. I did not know about Foxglove and that there is an almost identical alternative to Webviz.

jtbandes, 2022-03-08:
If you'd like to learn more about the differences between Foxglove Studio and Webviz, we have a blog post about it: https://foxglove.dev/blog/foxglove-studio-vs-webviz

mamacker, 2022-03-25:
We're going to be bringing Ouster Lidar into FoxGlove - just adding a comment to reinforce this need. :)

Grisly00, 2022-06-23:
Yes please!

defunctzombie, 2022-06-23:
Is this a separate request from supporting point clouds? We have support for point cloud messages in Studio which you can produce from many different types of lidar sensors. We've no plans to add custom support for ouster lidar beyond point clouds right now.

closed 2022-06-23 by jtbandes