foxglove/studio #782: Visualize URDF in 3D view

jtbandes, 2021-04-28:
We currently have a separate URDF Viewer panel, but users naturally want to visualize the robot model along with other markers, pointclouds, map, etc.

v4hn, 2021-06-15:
This is one of the basic requirements for everything involving manipulation. So yes, not having this is a deal-breaker for us.

defunctzombie, 2021-07-22:
Related idea (or part of the implementation) is a resource browser that would let users upload their resources and reference them in markers similar to how ROS markers allow for a mesh_resource string http://wiki.ros.org/rviz/DisplayTypes/Marker#Mesh_Resource_.28MESH_RESOURCE.3D10.29_.5B1.1.2B-.5D

jhurliman, 2021-10-15:
In https://github.com/foxglove/studio/pull/1983 we are adding support for `MESH_RESOURCE` markers, but currently only supporting HTTP(S) URLs pointing at glTF (.glb) 3D models. Support for the rviz-supported mesh formats (.stl and .dae) and `package://` URLs is planned.

amacneil, 2021-11-01:
Closing. Remaining work tracked in https://github.com/foxglove/studio/issues/2059

closed 2021-11-01 by amacneil