foxglove/studio #783: SSH tunnel connections

amacneil, 2021-04-28:
Similar to how vscode can work with a remote host over SSH, it would be nice if we supported connecting to ROS over an SSH tunnel. This would avoid many networking issues that come from all the various TCP connections ROS opens.

It would also mean that rosbridge (websocket) connections are likely no longer necessary for desktop users. However, we may wish to keep websocket support for browser users.

pschafhalter, 2021-07-28:
Hi, we would find this feature very valuable for our use case with [Pylot](https://github.com/erdos-project/pylot/), an open-source AV platform.

Typically, we deploy Pylot inside a docker container on a beefy remote machine. We SSH into the container to run Pylot. While we currently have a Pygame-based visualization that renders over X-forwarding,  we're currently working on moving our visualization stack to Foxglove in the following PR: erdos-project/pylot#201.

For now, we expect to use rosbridge and SSH port forwarding to visualize with Foxglove; however, connecting from Foxglove directly via SSH would be smoother.

Thanks for making this awesome software!

CC @icgog @storm-light

jtbandes, 2021-08-02:
Hi @pschafhalter, thanks for sharing your use case, the project looks very cool!

Is it true that Pylot doesn't use ROS, but you're adding ROS publishers just for the Foxglove Studio integration? Could you share some more about how the existing data logging and IPC works? This ticket was still assuming a ROS system, but I'd be curious to learn more about non-ROS use cases (see also: https://github.com/foxglove/studio/discussions/1268).

pschafhalter, 2021-08-02:
Hi @jtbandes,

Yes, that's true. Our current IPC stack uses TCP and Rust serialization libraries. It's currently not based on a standard format and relies on type checking for deserialization, but this is something we might look into in the future.

XDGFX, 2021-08-12:
This could be very useful for me also.

Currently it would be a nice alternative to running the ros bridge node, but down the line we're planning on having many robots on various networks across the country, connected with a mesh network (likely https://github.com/slackhq/nebula). It would be easy to ssh into a particular robot, which would be headless, for debugging and checking.

amacneil, 2022-02-09:
@pschafhalter FYI you should check out our new [Foxglove websocket protocol](https://github.com/foxglove/ws-protocol) (better suited for non-ROS middlewares).

While this feature request is easily worked around by tunneling the ws-protocol over SSH, it would be nice for us to eventually support direct SSH connections. So let's keep this ticket open.

pschafhalter, 2022-02-09:
Great, we will keep an eye on that! That might be a better solution for us as well, as let's us avoid depending on ROS.