foxglove/studio #800: Map tiles should match dark theme

jtbandes, 2021-04-29:
Consider using a dark tile set such as https://openmaptiles.org/styles/#dark-matter

defunctzombie, 2021-04-29:
This dark map has really bad contrast - I find it hard to read and would not want this to be the default map view without a way for the user to go back to a light map.

jtbandes, 2021-10-29:
Reopening the issue because I still want this feature. Regardless of whether that specific tile set is good, I think it's a reasonable request for the dark theme to use a dark tile set.

amacneil, 2021-10-29:
https://github.com/foxglove/studio/issues/2046

jtbandes, 2021-10-29:
Unrelated, but SGTM :)

jhurliman, 2021-12-14:
I think we will need to register for an API key? https://api.maptiler.com/maps/darkmatter/style.json?key=get_your_own_key_rSw2Lu595oi7U6WngsFQ

defunctzombie, 2022-01-13:
Removing myself. I am not actively working on this feature. I've commented on the referenced PR with some outstanding issues.

defunctzombie, 2022-05-13:
https://github.com/foxglove/studio/pull/3364 Allows users to select their own tile theme. I think if we want to do this we can add light and dark options to the map tile layer dropdowns and default to match the theme. This lets users easily override if they prefer a different contrast.

jtbandes, 2022-12-13:
Is there any hope of completing this issue? Map panel sticks out as one that does not really "support" dark mode by default. What would it take to add dark tiles to the dropdown?

defunctzombie, 2022-12-13:
Yea - the path forward here is to add a dark tile option to the dropdown list (before we didn't have an easy way to select different tiles via a dropdown but we do now). I still think that this is blocked on finding a dataset that is widely usable without random IP restrictions. I also feel like we have added the option to bring your own tile set so if you really want to use something else you can.