foxglove/studio #801: Hovering playback bar should highlight points in Map panel

jtbandes, 2021-04-29:
The currently hovered time `hoverValue` should be used in Map to show the closest GPS position when hovering on the playback bar (or on a Plot).

This code will probably need to be cleaned up to store a Time instead of an arbitrary "hoverValue" since the Map does not use a linear translation for highlighting the current position.

https://github.com/foxglove/studio/blob/6579ae338deb916fe4e665df62fe5845bee16df8/app/components/TimeBasedChart/HoverBar.tsx#L56

https://github.com/foxglove/studio/blob/6579ae338deb916fe4e665df62fe5845bee16df8/app/components/TimeBasedChart/index.tsx#L768-L771

As a bonus, hovering on the Map could also update the hoverValue and clicking can seek playback.

https://github.com/foxglove/studio/blob/fc858b1fd337f7413c6f6457befed78f35d66a91/app/panels/Plot/index.tsx#L265-L276

defunctzombie, 2021-04-29:
We can add a `const { hoverTime, setHoverTime } = useHoverTime()` to the panels api. Should also consider alternatives to calling it _hover_ tho maybe that is appropriate.

jtbandes, 2021-04-30:
Yeah, maybe something like `previewedTime`?

commit c57aac8f (2021-06-14): Add previewTime and setPreviewTime to studio extension api PanelContext

The previewTime indicates the time in seconds when a user is hovering
the cursor on the seek bar. The time is within the start/end range of playback.

The setPreviewTime function allows panels to manually set the preview time.

This change adds support for previewTime to the map panel. The map panel
will highlight the most recent gps coordinate leading up to the previewTime.

Fixes #801
PR #1200 (2021-06-14): Add previewTime and setPreviewTime to studio extension api PanelContext
The previewTime indicates the time in seconds when a user is hovering
the cursor on the seek bar. The time is within the start/end range of playback.

The setPreviewTime function allows panels to manually set the preview time.

This change adds support for previewTime to the map panel. The map panel
will highlight the most recent gps coordinate leading up to the previewTime.

Fixes #801

## User Facing changes

- Hovering the seek bar will highlight the relevant GPS coordinate
- Studio Extensions can watch previewTime and  [... cut]
commit 178d5c59 (2021-06-14): Add previewTime and setPreviewTime to studio extension api PanelContext (#1200)

The previewTime indicates the time in seconds when a user is hovering
the cursor on the seek bar. The time is within the start/end range of playback.

The setPreviewTime function allows panels to manually set the preview time.

This change adds support for previewTime to the map panel. The map panel
will highlight the most recent gps coordinate leading up to the previewTime.

Fixes #801
closed 2021-06-14 by defunctzombie