foxglove/studio #802: Content Security Policy of your site blocks the use of 'eval' in JavaScript

jhurliman, 2021-04-29:
The Content Security Policy (CSP) prevents the evaluation of arbitrary strings as JavaScript to make it more difficult for an attacker to inject unathorized code on your site.

To solve this issue, avoid using eval(), new Function(), setTimeout([string], ...) and setInterval([string], ...) for evaluating strings.
If you absolutely must: you can enable string evaluation by adding unsafe-eval as an allowed source in a script-src directive.
⚠️ Allowing string evaluation comes at the risk of inline script injection.

```
Source Location | Directive | Status
-- | -- | --
file:412466 | script-src | blocked
file:422950 | script-src | blocked
```

amacneil, 2021-04-29:
Happening for me on macOS too:

[image]

The strange thing is that we are explicitly allowing `unsafe-eval` in our content security policy:

https://github.com/foxglove/studio/blob/v0.6.0/desktop/index.ts#L173-L181

amacneil, 2021-04-29:
@jhurliman are you seeing any actual problems as a result of this or just the warning?

jhurliman, 2021-04-29:
@amacneil I haven't found the block(s) of code that are being disabled, it's possible they are unused and we can just delete some code to fix this. But I know we use `new Function()` for bayer decoding in the image panel and I haven't tested that yet.

amacneil, 2021-04-29:
We are definitely not blocking `eval` everywhere today. If you remove that from our CSP and run a production build there are a lot of devtools errors and the app refuses to load.

So, I think this error is coming from somewhere else.

amacneil, 2021-04-29:
I did some more testing of this and it feels like probably an electron bug (or undocumented behavior).

- Removing `unsafe-inline` from CSP causes our app to definitely not work.
- Opening up our CSP to be as [flexible as possible](https://stackoverflow.com/a/51067035/462702) still causes this error in devtools, although no noticeable problems in the app besides the error showing up.
- Moving our CSP to a `<meta http-equiv="">` tag still shows the error.
- Removing our CSP from every request except `renderer/index.html` still shows the error.
- Removing our CSP entirely fixes the error

Basically, I can't find any combination of CSP that avoids this error, other than no CSP at all. However, the error doesn't seem to point to any specific file that failed to load, and I can't see any negative side effects in the app.

defunctzombie, 2021-05-28:
I've removed the bug label. We've not identified anything that is broken as a result of this. Seems like an erroneous warning within electron.

defunctzombie, 2021-05-28:
I no longer see this in the issues tab on main (on dev or prod build).

closed 2021-05-28 by defunctzombie