foxglove/studio #803: Production app tries to load preload.js.map from the wrong folder

jhurliman, 2021-04-29:
```
DevTools failed to load SourceMap: Could not parse content for file:///C:/Users/jhurl/AppData/Local/Programs/foxglove-studio/resources/app.asar/renderer/preload.js.map: Unexpected end of JSON input
```

If you open up the ASAR archive, that file is actually at `app.asar/main/preload.js.map`. It's unclear why it is using the `app.asar/renderer/` prefix instead.

amacneil, 2021-04-29:
Which version & OS? I'm not seeing this on 0.6.0/macOS.

The only source map warning I see is for `renderer/module.wasm.map`.

jhurliman, 2021-04-30:
0.6.0 production on Windows

defunctzombie, 2021-05-12:
Is this still an issue now since we send source maps to sentry during the webpack build?

defunctzombie, 2021-05-14:
I don't see this in 0.8.1

closed 2021-05-14 by defunctzombie