foxglove/studio #846: Provider topics should not change often

amacneil, 2021-05-03:
> If they do they are probably not memoized properly. Please let the Foxglove team know if you see this warning.

This happens when connecting to a websocket (just simple rosbridge+roscore running inside a VM) on latest main.

Not sure what it means but it is not actionable so should probably not be displayed to the user at all.

[image]

defunctzombie, 2021-05-14:
@jtbandes @amacneil I researched this further to understand why we get this. I wasn't able to reproduce the warning by connecting immediately to websocket but I was able to reproduce by having a local bag file open and then switching to a websocket connection.

The warning comes from https://github.com/foxglove/studio/blob/main/app/components/MessagePipeline/index.tsx#L245 which uses https://github.com/foxglove/studio/blob/f4da0a890d275c0619e7477005507dd649001d72/app/hooks/useShouldNotChangeOften.ts#L17. This hook checks that the value has not changed 3 times within 200 ms. However, when changing the player the value _does_ change 3 times in 200ms. The current player topics value, then undefined while websocket is connecting, and finally the new topics value.

I am not convinced on the usefulness of this warning; It is definitely a false positive in the situation I described.

amacneil, 2021-05-17:
Maybe we just remove this message then? We see this a lot in sentry too. I'm unclear whether there is any real issue for us to be concerned about here.

amacneil, 2021-05-18:
@defunctzombie is this closed by https://github.com/foxglove/studio/pull/994?

defunctzombie, 2021-05-18:
@amacneil no - it will be fixed by https://github.com/foxglove/studio/pull/993

commit fbe63401 (2021-05-17): Memo player activeData.topics with useShallowMemo in MessagePipeline

Rather than warning that player topics are changing frequently which
produces false positives during player changes, useShallowMemo to keep
a stable reference to topics as long as the list of topics within the
array is unchanged.

Fixes #846
PR #993 (2021-05-18): Memo player activeData.topics with useShallowMemo in MessagePipeline
Rather than warning that player topics are changing frequently which
produces false positives during player changes, useShallowMemo to keep
a stable reference to topics as long as the list of topics within the
array is unchanged.

Fixes #846
commit 4d414b35 (2021-05-18): Memo player activeData.topics with useShallowMemo in MessagePipeline (#993)

Rather than warning that player topics are changing frequently which
produces false positives during player changes, useShallowMemo to keep
a stable reference to topics as long as the list of topics within the
array is unchanged.

Fixes #846
closed 2021-05-18 by defunctzombie