foxglove/studio #849: JavaScript Math Suffixes for Message Path Syntax Only Work in Plot Panel

esthersweon, 2021-05-03:
JavaScript Math suffixes currently only work in the Plot, even though other panels (like Raw Messages) does not complain when a topic message path using a JavaScript Math suffix is used

These suffixes should probably work everywhere message path syntax is used.

[image]

jtbandes, 2021-05-25:
http://wiki.ros.org/Names

https://github.com/ros/ros_comm/blob/f5fa3a168760d62e9693f10dcb9adfffc6132d22/tools/rosgraph/src/rosgraph/names.py#L216

esthersweon, 2021-06-10:
Addressed in https://github.com/foxglove/studio/pull/1194.