foxglove/studio #879: Ability to publish clicks from Image panel

amacneil, 2021-05-05:
We should support clicking on the image panel, and publishing these clicks on a topic.

Open questions:
- Which message type should we use? @jhurliman suggested simple `geometry_msgs/Point`.
- Do we publish each click as one message, or some frequency?
- Should we also support clicking on markers, and republishing the selected marker?

jtbandes, 2021-05-05:
Related to https://github.com/foxglove/studio/issues/605

elephantum, 2021-05-06:
For me, as a potential user of the feature:

- `geometry_msgs/Point` seems fine
- each click as a message would be more flexible, they could be aggregated downstream if needed

amacneil, 2021-05-06:
Blocked by https://github.com/foxglove/studio/issues/697 since we currently only support publishing via websocket connections.

rgov, 2021-05-23:
Points would be great, and it would be excellent if I could drag a rectangle as well. My use case is selecting a ROI to track. I could have the user select two points but the UI feedback is tricky.

amacneil, 2021-11-20:
Valid use case since it is possible in rqt.

If folks are interested in seeing this feature land please feel free to add comments/upvotes.

jbcpollak, 2022-03-14:
this would be useful to us, feature as described by @elephantum would be perfect as a minimum to get started - plenty of ways to make it more advanced in the future.