foxglove/studio #893: `yarn serve` doesn't watch all files

jtbandes, 2021-05-07:
Steps to reproduce:
- Run `yarn serve` on main.
- Edit the `PanelConfigSchemaEntry` type in `app/types/panels.ts`, or edit the `RegisteredIconNames` type in `app/typings/fluentui.d.ts`.

Webpack does not even attempt to rebuild any files when these two files are changed.

It seems likely that this is a regression caused by #829 / #839.

jtbandes, 2021-05-11:
I created a minimal repro and filed an issue against fork-ts-checker: https://github.com/TypeStrong/fork-ts-checker-webpack-plugin/issues/611

defunctzombie, 2021-05-14:
@jtbandes I propose removing the _bug_ label from this issue - maybe a different label? I'd like to keep bug as an indicator for user-facing/product bugs.

jtbandes, 2022-09-08:
Seems like this has improved.

closed 2022-09-08 by jtbandes