foxglove/studio #919: "Cannot load two HTML5 backends at the same time"

wkalt, 2021-05-12:
I moved foxglove studio between two monitors and got this error immediately. Here's the full stack trace:
```
Error: Cannot have two HTML5 backends at the same time.
    at HTML5BackendImpl.setup (webpack-internal:///../../node_modules/react-dnd-html5-backend/dist/esm/HTML5BackendImpl.js:432:15)
    at DragDropManagerImpl.handleRefCountChange (webpack-internal:///../../node_modules/dnd-core/dist/esm/classes/DragDropManagerImpl.js:26:25)
    at Object.dispatch (webpack-internal:///../../node_modules/redux/es/redux.js:307:7)
    at HandlerRegistryImpl.addSource (webpack-internal:///../../node_modules/dnd-core/dist/esm/classes/HandlerRegistryImpl.js:102:18)
    at registerSource (webpack-internal:///../../node_modules/react-dnd/dist/cjs/internals/registration.js:19:27)
    at registerDragSource (webpack-internal:///../../node_modules/react-dnd/dist/cjs/hooks/useDrag/useRegisteredDragSource.js:36:59)
    at commitHookEffectListMount (webpack-internal:///../../node_modules/react-dom/cjs/react-dom.development.js:20568:26)
    at commitLifeCycles (webpack-internal:///../../node_modules/react-dom/cjs/react-dom.development.js:20629:11)
    at commitLayoutEffects (webpack-internal:///../../node_modules/react-dom/cjs/react-dom.development.js:23421:7)
    at HTMLUnknownElement.callCallback (webpack-internal:///../../node_modules/react-dom/cjs/react-dom.development.js:3945:14)
Component stack:

    at Panel(undefined) (webpack-internal:///../../app/components/Panel.tsx:116:17)
    at div
    at div
    at InternalMosaicWindow (webpack-internal:///../../node_modules/react-mosaic-component/lib/MosaicWindow.js:53:47)
    at DragSource(InternalMosaicWindow) (webpack-internal:///../../node_modules/react-dnd/dist/cjs/decorators/decorateHandler.js:83:7)
    at DropTarget(DragSource(InternalMosaicWindow)) (webpack-internal:///../../node_modules/react-dnd/dist/cjs/decorators/decorateHandler.js:83:7)
    at MosaicWindow (webpack-internal:///../../node_modules/react-mosaic-component/lib/MosaicWindow.js:241:42)
    at div
    at div
    at MosaicRoot (webpack-internal:///../../node_modules/react-mosaic-component/lib/MosaicRoot.js:42:47)
    at div
    at MosaicWithoutDragDropContext (webpack-internal:///../../node_modules/react-mosaic-component/lib/Mosaic.js:55:28)
    at ErrorBoundary (webpack-internal:///../../app/components/ErrorBoundary.tsx:58:9)
    at UnconnectedPanelLayout (webpack-internal:///../../app/components/PanelLayout.tsx:99:13)
    at PanelLayout (webpack-internal:///../../app/components/PanelLayout.tsx:146:12)
    at div
    at Stack (webpack-internal:///../../node_modules/@fluentui/foundation-legacy/lib/createComponent.js:42:99)
    at ErrorBoundary (webpack-internal:///../../app/components/ErrorBoundary.tsx:58:9)
    at div
    at div
    at MosaicRoot (webpack-internal:///../../node_modules/react-mosaic-component/lib/MosaicRoot.js:42:47)
    at div
    at MosaicWithoutDragDropContext (webpack-internal:///../../node_modules/react-mosaic-component/lib/Mosaic.js:55:28)
    at div
    at HideRootDropTargets__components_Sidebar_index_tsx (webpack-internal:///../../node_modules/styled-components/dist/styled-components.browser.cjs.js:2:18799)
    at div
    at Stack (webpack-internal:///../../node_modules/@fluentui/foundation-legacy/lib/createComponent.js:42:99)
    at Sidebar (webpack-internal:///../../app/components/Sidebar/index.tsx:59:20)
    at div
    at MultiProvider (webpack-internal:///../../app/components/MultiProvider.tsx:19:26)
    at Workspace (webpack-internal:///../../app/Workspace.tsx:173:71)
    at BuiltinPanelCatalogProvider (webpack-internal:///../../app/context/BuiltinPanelCatalogProvider.tsx:26:72)
    at Suspense
    at DndProvider (webpack-internal:///../../node_modules/react-dnd/dist/cjs/core/DndProvider.js:39:23)
    at NativeAppMenuProvider (webpack-internal:///./components/NativeAppMenuProvider.tsx:26:65)
    at AssetsProvider (webpack-internal:///../../app/context/AssetContext.tsx:38:27)
    at Provider (webpack-internal:///../../app/util/createSelectableContext.tsx:33:25)
    at MessagePipelineProvider (webpack-internal:///../../app/components/MessagePipeline/index.tsx:71:36)
    at PlayerManager (webpack-internal:///../../app/components/PlayerManager.tsx:294:26)
    at Connect(PlayerManager) (webpack-internal:///../../node_modules/react-redux/es/components/connectAdvanced.js:231:68)
    at AnalyticsProvider (webpack-internal:///../../app/context/AnalyticsProvider.tsx:30:148)
    at Provider (webpack-internal:///../../node_modules/react-redux/es/components/Provider.js:18:20)
    at ToastProvider (webpack-internal:///../../node_modules/react-toast-notifications/dist/ToastProvider.js:61:5)
    at StudioToastProvider (webpack-internal:///../../app/components/StudioToastProvider.tsx:41:242)
    at ModalHost (webpack-internal:///../../app/context/ModalHost.tsx:22:22)
    at exports.ThemeProvider (webpack-internal:///../../node_modules/styled-components/dist/styled-components.browser.cjs.js:2:23926)
    at ThemeContainer (webpack-internal:///../../app/theme/ThemeProvider.tsx:100:204)
    at ThemeProvider (webpack-internal:///../../node_modules/@fluentui/react/lib/utilities/ThemeProvider/ThemeProvider.js:19:128)
    at ThemeProvider (webpack-internal:///../../app/theme/ThemeProvider.tsx:105:26)
    at OsContextLayoutStorageProvider (webpack-internal:///../../app/context/OsContextLayoutStorageProvider.tsx:29:68)
    at OsContextAppConfigurationProvider (webpack-internal:///../../app/context/OsContextAppConfigurationProvider.tsx:27:46)
    at MultiProvider (webpack-internal:///../../app/components/MultiProvider.tsx:19:26)
    at ErrorBoundary (webpack-internal:///../../app/components/ErrorBoundary.tsx:58:9)
    at App (webpack-internal:///./App.tsx:101:90)
```

jtbandes, 2021-05-12:
I often see this when a code changes causes webpack to do a hot reload. Are you sure this is triggered just by moving between monitors? (Does it happen in a prod build?)

defunctzombie, 2021-05-17:
Using latest main I dragged a window between desktops on my mac and there was no such crash. @wkalt could you provide more details on platform/reproduction steps?

wkalt, 2021-05-28:
I'm unable to reproduce this.

closed 2021-05-28 by wkalt