foxglove/studio #959: Data Source Info panel eats an unnecessary amount of CPU

jhurliman, 2021-05-15:
The `Data Source Info` panel is refreshing at full framerate (60fps) and due to the timestamps, it redraws every frame at a cost of 1-1.5ms per frame on my machine. For me this translates into ~20% CPU usage and "Very High" power usage from our app. When I change that panel to something else, CPU usage drops to <5% and power usage goes to "Moderate".

This could probably be fixed by throttling updates to 5-10Hz, there's no reason that panel needs to update so quickly.

defunctzombie, 2021-05-16:
This surfaces an important question. Should players (or message pipeline) throttle down the message stream itself? Maybe to 30hz even? We aren't aiming for game engine refresh rates so that seems sensible when playing back live?

jtbandes, 2021-05-16:
> We aren't aiming for game engine refresh rates

Why not?

defunctzombie, 2021-05-16:
> Why not?

Is 60 better than 30 in a noticeable way for playback? Guess it depends who you ask :/

jtbandes, 2021-05-17:
I suppose latency is more noticeable than framerate, but do think the difference between 30 and 60 is noticeable. Especially if you are trying to look at something that's flickering.

jhurliman, 2021-05-21:
For users reading this issue, it can now be mitigated by reducing the "Message rate (Hz)" setting (click the settings cog in the bottom-right corner).

I'm leaving this ticket open so we can address the root cause of expensive re-renders in the Data Source Info panel. I have a suspicion this affects the Parameters panel too, but that panel redraws much less often.

defunctzombie, 2021-05-28:
@jhurliman is this during bag playback or during live playback?

amacneil, 2021-12-09:
@2metres @defunctzombie we should address this as part of moving this info to the Data Source sidebar (or at least ensure that the DS sidebar doesn't suffer from the same issue)

defunctzombie, 2021-12-09:
Yea - One idea here is that we should move all the components that need to render every frame "end time is a culprit here" into a nested component so it doesn't render everything else. Maybe that isn't true and the whole display renders ever frame (cause of the Hz readings). I think we could drop those from sidebar display and focus on the more static data. If you want to track Hz we can do that in a panel or some other data source viewer since it won't be a thing for live sources.

jtbandes, 2021-12-10:
FYI the Hz readings are not computed dynamically. They are just `topic.numMessages / duration`.

jhurliman, 2022-05-13:
This has been resolved by moving all the Data Source Info panel data into the connection sidebar, making this panel redundant.

closed 2022-05-13 by jhurliman