foxglove/studio #974: Allow overriding the default theme in ThemeProvider

defunctzombie, 2021-05-15:
Add the ability for a user of <ThemeProvider> to override the theme with their own.

jtbandes, 2021-05-17:
By "a user", do you mean a developer or an end user?

defunctzombie, 2021-05-17:
Eventually both - but this issue is specifically by a developer. Rather than hard-code the theme in the theme provider the theme provider accepts a _value_. The thinking was to allow theming within the environment by a developer - but maybe this ticket is better focused on allowing _end user_ theming.

jtbandes, 2022-01-05:
I think now that we have dark and light mode, the groundwork is laid for any further theme customizability. Any further development would need to be driven by specific feature requests.

closed 2022-01-05 by jtbandes