foxglove/studio #996: Migrate Data Source Info panel into Connection sidebar

amacneil, 2021-05-17:
Now that we have the Connection sidebar, it would be a logical place to show information about the current connection.

I propose removing the Data Source Info panel, and surfacing this information in the Connection sidebar for the active connection. It might need some layout adjustment to make sense given the sidebar width.

[image]

amacneil, 2021-11-14:
Still think we should do this

2metres, 2022-03-22:
[image]

PR #3060 (2022-03-31): Move data source info into Data Source Sidebar
**User-Facing Changes**
Move data source info into Data Source Sidebar once connection is live.

**Description**

https://user-images.githubusercontent.com/924528/159839105-24495525-bfa1-42fc-a9db-9647bea56d5c.mov

Closes #996
closed 2022-03-31 by 2metres