foxglove/studio #1036: Microsoft Defender SmartScreen

daniel-bogdoll, 2021-05-20:
Hey everybody,

the installation of v0.9.0 is being "protected" by Microsoft Defender SmartScreen on Windows 10 Pro, version 20H2. Installation is only possible if you click on "Furher information" (Weitere Informationen), which gives you th option to "install anyway" (Trotzdem ausführen).

[image]
[image]

amacneil, 2021-05-20:
Thanks for the reminder! I was hoping that with enough installs this will go away. 

I guess we will have to buy the physical EV code signing key that avoids this warning altogether. It means our CI won’t be able to sign the builds, but at least will prevent the warning.

amacneil, 2021-07-19:
I am going to close this because the only immediate solution (physical EV code signing token) would break our ability to do release builds via CI. I expect that after some months Windows Defender will start trusting our existing signing certificate. If this doesn't happen in the next few months we can reconsider.

In the meantime, please "install anyway" so that Windows can learn to trust us.

amacneil, 2022-01-04:
Reopening because I think these warnings are still happening and we should find a solution to address them.

amacneil, 2022-02-24:
EV code signing certs require a hardware token or HSM. It seems like not all cert providers support HSMs.

Prior art for code signing:

- https://www.ssl.com/guide/supported-cloud-hsms-document-signing-ev-code-signing/
- https://github.com/nextgens/authenticode-sign-action (probably don't use directly but we could borrow ideas)
- https://www.electron.build/tutorials/code-signing-windows-apps-on-unix

Also we could run a local server with a hardware token.

amacneil, 2022-03-30:
@defunctzombie has saved us from this pain once and for all - fixed in v1.5.1 release.

closed 2022-03-30 by amacneil