foxglove/studio #1037: deb file problem

thoys, 2021-05-20:
while upgrading the apt-get package on Ubuntu (through WSL), I came across the following issue:

```
E: Failed to fetch https://apt.foxglove.dev/studio/pool/main/f/foxglove-studio/foxglove-studio-0.9.0-linux-amd64.deb  File has unexpected size (133 != 73922060). Mirror sync in progress? [IP: 172.67.189.144 443]
   Hashes of expected file:
    - SHA512:d18bceb9e9b623111ff823cc02ac99cc6c700858cf658cde3ae511f8460f24eb89d86a910c35fa64e97718d9ef5c55ffaa6f89b552c438b285752c7de70f1fd2
    - SHA256:23dada2a736cec68bc96cf71b8e2a9765e93af114f29bfd000fd3b0c005cd59e
    - SHA1:b8fe038e226928f853514982b54089a75285f8cc [weak]
    - MD5Sum:8e8aecceda4b03b0c8c1f376513a7881 [weak]
    - Filesize:73922060 [weak]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

While manually downloading the deb-file and confirming it really was 133bytes I discovered that its contents were merely a link to a git-lfs file:
```
version https://git-lfs.github.com/spec/v1
oid sha256:23dada2a736cec68bc96cf71b8e2a9765e93af114f29bfd000fd3b0c005cd59e
size 73922060
```

I expect things didn't copy correctly in the publish process. Maybe the system that deals with the publishing of the packages did not have lfs support?

defunctzombie, 2021-05-20:
@thoys Have you been able to apt-get upgrade this package prior to v0.9.0? I'm hoping to narrow down if this is a new issue or a recent development.

defunctzombie, 2021-05-20:
@thoys I've re-published the deb files and did a manual download of the URL above to confirm a full deb is downloaded (not a git-lfs link) - when you get a chance please try to apt-get upgrade the package again. Feel free to close the issue if you are able to successfully upgrade or add additional comments if anything else is not working as expected.

thoys, 2021-05-20:
heya @defunctzombie thanks. this worked perfectly thanks!

closed 2021-05-20 by thoys