foxglove/studio #1047: Disable macOS code signing during packaging by default (except in CI)

jhurliman, 2021-05-20:


defunctzombie, 2021-06-18:
I think this happens when you have an apple developer identity installed but not the one electron-builder is expecting to sign the app.

amacneil, 2021-06-25:
I don't actually have a way to repro this since my macOS device has the correct cert installed. It seems possible that we could configure electron-builder to only sign when a particular env var is present, but I don't see any obvious way to configure that behavior.

jtbandes, 2021-06-25:
I think the example given by a user was not another Apple developer identity, but some other random gpg signing key that showed up when running `security find-identity -v -p codesigning`

amacneil, 2021-07-19:
This hasn't come up for a while, and I can't see any obvious way to disable it in electron-builder. Recommend closing.

defunctzombie, 2021-07-19:
I think this is already disable if you don't have a signing identity so without further details/reproduction/desired workflow there is no further action here.

closed 2021-07-19 by defunctzombie