foxglove/studio #1049: Further customization of geometry_msgs/PoseStamped

jhurliman, 2021-05-20:
From https://github.com/cruise-automation/webviz/issues/586.

> I would like to visualize messages of the type geometry_msgs/PoseStamped using a rendered Arrow, but the arrow length is always defaulted to 2.0m. I I would like to have much smaller arrows (centimeter-long) for better visualization of my application. The only portions of the arrow that are configurable are the head and its width. It would be great if we could also configure the Shaft Length (see below for the available configurations for geometry_msgs/PoseStamped arrow)

[image]

amacneil, 2021-11-19:
While we're here, let's change the arrow default dimensions to actually be an arrow and not a cylinder.

jhurliman, 2022-07-14:
Implemented in New3d

closed 2022-07-14 by jhurliman