foxglove/studio #1069: API for extending the Image panel with new formats and layers

rgov, 2021-05-24:
I have implemented a Theora video decoder (#537) as a separate panel, but since it is not derived from the Image panel, it lacks pan, zoom, marker overlays, hotkeys, etc. It would be useful to have a way to extend the Image panel so that I get these features.

Consider an API through which a plugin could specify the message types handled and a hook that returns a `ImageBitmapSource` to be displayed.

esthersweon, 2021-05-25:
In the interim, to unblock @rgov – I pushed a template commit for adding this decoding functionality directly into the Image panel here: https://github.com/foxglove/studio/commit/a0b6f614db5957d59c1e3dd532a865a0e03fa4cb.

rgov, 2021-12-09:
I think this one should be considered.

jtbandes, 2022-12-13:
Will note that we recently shipped [message converter extensions](https://foxglove.dev/blog/announcing-message-converter-extensions-in-foxglove-studio) which provide a way to convert custom formats to a format supported by Studio (such as [`foxglove.RawImage`](https://foxglove.dev/docs/studio/messages/raw-image)).