foxglove/studio #1070: Request: Example panels for integrating other visualization libraries

rgov, 2021-05-24:
It may be necessary for a user to implement a custom panel to visualize certain message types for which the built-in plot panels are insufficient. For instance, Chart.js doesn't support polar plots, which are needed in one project of mine.

However, implementing a panel is complicated for users who are more familiar with writing ROS nodes than React components.

Therefore it would be useful to have examples of how to integrate other common visualization libraries (D3, Observable Plot, Vega-Lite, or P5.js) and allow the developer to focus on just modifying the plotting code.

defunctzombie, 2021-06-01:
We are working on extension support now and will provide some example extensions that will _hopefully_ provide some guidance on how to implement basic panels. We are also considering lifting the react requirement for extensions - tho we think react is a good way to handle UI code it also isn't for everyone's needs.

amacneil, 2021-06-01:
In addition to extensions, I think "powerful out of the box plots" is something that we aspire to support as a core feature. As you noted, most robotics devs shouldn't have to think about tradeoffs between JS chart libs.

Regardless, I agree with the original comment in this ticket ("better documentation for panel extensions"). I'm not sure we need to keep this ticket open though since we haven't officially launched extension support yet, and documentation is absolutely a priority before we launch it.

rgov, 2021-11-29:
There is (last I checked) a sample that does not use React and I didn't find it too difficult to import D3 and use it.

closed 2021-11-29 by rgov