foxglove/studio #1071: Request: Read-only mode

rgov, 2021-05-24:
Studio is useful for creating dashboards that show real-time sensor information. However, in some circumstances it could be desirable to have the panel arrangement "locked," such as on system where non-expert users might be looking at data but should not be reconfiguring Studio.

I'm not suggesting anything super secure here. It would probably be enough to have a menu item for locking the layouts. Add an overlay element across the whole panel area that intercepts clicks, but allows mouseover.

rgov, 2021-12-15:
> * Would you want to lock just the layout tile positions, or just the settings within each panel, or both?

I think both. Once we have dialed in a good layout I don't want anyone accidentally dragging a panel somewhere, changing a plot line, etc. and needing to know how to reset it.

> * Are you expecting this to be used for viewing data only, or would you likely have one or more publish panels to control the robot somehow? (e.g. just a dashboard vs an operator console)

I could see use cases for having publish panels in a locked layout.

> * If we implemented publishing support in other panels (such as [Ability to publish clicks from Image panel #879](https://github.com/foxglove/studio/issues/879)), would you want these to be disabled too, or would you still want those features to work for the operator?

If the setting was turned on before the layout was locked, and other panels are able to publish, I would expect this to work as well. I don't have an immediate use case for this feature though.

> * Would a desktop app still work in this instance? Or would you want to combine this with a web-based version? (e.g. "export as dashboard", or maybe some way to drop a configuration file alongside a web build to make a static operator console)

I think in the use cases I imagine, most users would access via the browser. However the desktop version should also be able to lock layouts for field operations.

wouterio, 2022-12-09:
I'm interested in this feature too, see #4936 . Would it be enough if the buttons to add/remove/configure panels can be disabled?

wouterio, 2022-12-14:
> Would you want to lock just the layout tile positions, or just the settings within each panel, or both?

Users moving panels around would be no problem for me, as long as they can't add/remove/configure panels. But locking tile positions would be fine too.

> Are you expecting this to be used for viewing data only, or would you likely have one or more publish panels to control the robot somehow? (e.g. just a dashboard vs an operator console)

Depends on the use case. Instead of either dashboard mode or operator mode, I'd go for a more fine-grained approach where the developer can choose per panel whether it should be interactive or not. 

Let's say you equip each panel configuration with the option to enable/disable interaction. That gives the developer full control which panel should be interactive or not. Locking the user interface then disables access to the configuration, so that the user can only interact with panels that the developer choose to be interactive.

> If we implemented publishing support in other panels (such as Ability to publish clicks from Image panel #879), would you want these to be disabled too, or would you still want those features to work for the operator?

Like what I wrote above, and what @rgov wrote.

> Would a desktop app still work in this instance? Or would you want to combine this with a web-based version? (e.g. "export as dashboard", or maybe some way to drop a configuration file alongside a web build to make a static operator console)

Agree with @rgov.